openwrt/mt76 #966: 7621u: crash in AP mode

bopjiang, 2025-03-12:
I am setting up an AP hotspot using a Raspberry Pi 5 with a USB wireless dongle (7621u chipset). 

The system crashes when certain clients are connected. Connecting with an iPhone works fine, but when using a Linux laptop, it always crashes.

I can reproduce the problem on both SNAPSHOT r28965-371cad4f28 and 24.10.0.

crash log:
```
[  121.515854] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000400
[  121.524690] Mem abort info:
[  121.527487]   ESR = 0x0000000096000005
[  121.531245]   EC = 0x25: DABT (current EL), IL = 32 bits
[  121.536578]   SET = 0, FnV = 0
[  121.539637]   EA = 0, S1PTW = 0
[  121.542784]   FSC = 0x05: level 1 translation fault
[  121.547679] Data abort info:
[  121.550563]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
[  121.556069]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[  121.561136]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[  121.566468] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000103b40000
[  121.572932] [0000000000000400] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[  121.581673] Internal error: Oops: 0000000096000005 [#1] SMP
[  121.587265] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet brcmfmac_wcc(O) mt7921u(O) mt7921_common(O) brcmfmac(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt792x_usb(O) mt792x_lib(O) mt76_usb(O) mt76_connac_lib(O) mt76(O) mac80211(O) lzo cfg80211(O) usbhid slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress libcrc32c hid_generic crc_ccitt compat(O) brcmutil(O) snd_bcm2835(C) hid evdev pwm_fan snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore nls_utf8 cmac crypto_acompress nls_iso8859_1 nls_cp437 vfat fat
[  121.666883] CPU: 1 PID: 259 Comm: mt76-usb-rx phy Tainted: G         C O       6.6.80 #0
[  121.675007] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[  121.680859] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  121.687849] pc : mt792x_rx_get_wcid+0x3c/0xdc [mt792x_lib]
[  121.693364] lr : mt7921_queue_rx_skb+0x140/0x7ec [mt7921_common]
[  121.699398] sp : ffffffc08148bcf0
[  121.702718] x29: ffffffc08148bcf0 x28: 0000000000000000 x27: 0000000000000000
[  121.709884] x26: 0000000060000c4f x25: 00000000080328bc x24: 0000000038010145
[  121.717048] x23: ffffff8100e32000 x22: 0000000080843801 x21: ffffff8102f22020
[  121.724213] x20: ffffff8102672528 x19: ffffff8102672500 x18: 0000000000000070
[  121.731377] x17: ffffffc17e2a3000 x16: ffffffc080008000 x15: ffffff81fef51680
[  121.738541] x14: 0000000375b3cb85 x13: 0000138b86b386bf x12: 0000000000000001
[  121.745706] x11: 00000000b123f581 x10: 00000000000008b0 x9 : 0000000000000000
[  121.752871] x8 : ffffff81026725b8 x7 : 0000000000000000 x6 : 000000000000003f
[  121.760035] x5 : 0000000000000040 x4 : ffffffc07935dc58 x3 : 0000000000004760
[  121.767199] x2 : 0000000000000000 x1 : 000000000000000a x0 : 0000000000000000
[  121.774363] Call trace:
[  121.776811]  mt792x_rx_get_wcid+0x3c/0xdc [mt792x_lib]
[  121.781974]  mt76u_stop_rx+0xbbc/0xecc [mt76_usb]
[  121.786700]  __mt76_worker_fn+0x90/0xe0 [mt76]
[  121.791166]  kthread+0xdc/0xe0
[  121.794228]  ret_from_fork+0x10/0x20
[  121.797815] Code: 54000101 79417801 360800e1 f941bc00 (f9420000) 
[  121.803931] ---[ end trace 0000000000000000 ]---
[  121.808561] Kernel panic - not syncing: Oops: Fatal exception
[  121.814327] SMP: stopping secondary CPUs
[  121.818261] Kernel Offset: disabled
[  121.821756] CPU features: 0x1,00000000,70028041,0000700b
[  121.827085] Memory Limit: none
[  121.830144] Rebooting in 3 seconds..
```

system info:
```json
root@OpenWrt:~# ubus call system board
{
        "kernel": "6.6.80",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 1",
        "model": "Raspberry Pi 5 Model B Rev 1.0",
        "board_name": "raspberrypi,5-model-b",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "firmware_url": "https://downloads.openwrt.org/",
                "revision": "r28965-371cad4f28",
                "target": "bcm27xx/bcm2712",
                "description": "OpenWrt SNAPSHOT r28965-371cad4f28",
                "builddate": "1741610382"
        }
}
```

wifi status:
`radio0` is the Raspberry Pi built-in wireless.

```json
        "radio1": {
                "up": true,
                "pending": false,
                "autostart": true,
                "disabled": false,
                "retry_setup_failed": false,
                "config": {
                        "path": "platform/axi/1000120000.pcie/1f00300000.usb/xhci-hcd.1/usb4/4-1/4-1:1.0",
                        "band": "5g",
                        "channel": "auto",
                        "htmode": "HE80",
                        "disabled": false
                },
                "interfaces": [
                        {
                                "section": "default_radio1",
                                "ifname": "wlan1",
                                "config": {
                                        "ifname": "wlan1",
                                        "mode": "ap",
                                        "ssid": "XXXXXX",
                                        "encryption": "psk2",
                                        "key": "XXXXXX",
                                        "network": [
                                                "lan"
                                        ],
                                        "mode": "ap"
                                },
                                "vlans": [

                                ],
                                "stations": [

                                ]
                        }
                ]
        }
}

```

NadavTasher, 2025-03-15:
Damn.
I'm experiencing the exact same thing here - also have an MT7921, using a stable 24.10.0 image.

I have not been able to extract the full call trace as I do not have a serial connection to my device, but this is what I got by dumping to a log file:
```
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.010002] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000400
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.010789] Mem abort info:
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.011036]   ESR = 0x0000000096000006
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.011366]   EC = 0x25: DABT (current EL), IL = 32 bits
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.011831]   SET = 0, FnV = 0
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.012101]   EA = 0, S1PTW = 0
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.012378]   FSC = 0x06: level 2 translation fault
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.012806] Data abort info:
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.013060]   ISV = 0, ISS = 0x00000006, ISS2 = 0x00000000
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.013539]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.013980]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.014447] user pgtable: 4k pages, 48-bit VAs, pgdp=00000001f6cfb000
Sat Mar 15 01:58:27 2025 kern.alert kernel: [  122.015011] [0000000000000400] pgd=08000001f1b5d003, p4d=08000001f1b5d003, pud=08000001f34b5003, pmd=0000000000000000
Sat Mar 15 01:58:27 2025 kern.emerg kernel: [  122.016008] Internal error: Oops: 0000000096000006 [#1] SMP
Sat Mar 15 01:58:27 2025 kern.warn kernel: [  122.016521] Modules linked in: xt_connlimit pppoe ppp_async nft_fib_inet nf_flow_table_inet nf_conncount mt7921e(O) mt7921_common(O) iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connlabel xt_connbytes xt_REDIRECT xt_MASQUERADE xt_CT pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt792x_lib(O) mt76_connac_lib(O) mt76(O) mac80211(O) lzo ipt_REJECT cfg80211(O) xt_time xt_tcpudp xt_recent xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc r8169 nfnetlink nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 lzo_rle lzo_decompress lzo_compress iptable_mangle iptable_filter ip_tables crc_ccitt compat(O) ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun sha512_generic sha512_arm64 seqiv sha3_generic jitterentropy_rng
```

I have narrowed it down a little - this only happens when I connect a specific device to the AP, my device being a `Lenovo CT-X636F` Chromebook.

OpenWrt running on a `FriendlyElec NanoPi R5S` - MT7921 connected through PCI-e.

NadavTasher, 2025-03-15:
@LorenzoBianconi Any idea why this might happen?

robgssp, 2025-03-23:
[This kernel patch](https://mail.spinics.net/lists/stable/msg815117.html) appears to fix it.

NadavTasher, 2025-03-29:
Was this patch backported to 6.6?
Will it be applied to the OpenWrt kernel tree for the next OpenWrt dot release?

NadavTasher, 2025-03-30:
This does not look like the same bug.
The code segment changed by the patch you posted is incoherent with the kernel error I experienced.