openwrt/mt76 #970: MT76x2 issue after 5 days uptime on v24.10.0

alcatraz-zz, 2025-03-21:
Hey fellas

Got a Xiaomi MI-R3G with 5GHz wifi that stopped working after 5 days uptime. It's running the latest stable 24.10.0 version of OpenWRT.

mt76x2e 0000:02:00.0: MCU message 1f

This appeared in the log. Kernel error something.

Restarting/disable/enable 5ghz did not fix it. I had to restart the router to fix it.

Thanks for all your work.

lvent, 2026-01-11:
I have also been dealing with this myself on a Mi4A gigabit ( ever since I upgraded to 24.10.4 from 23.x something)
Changing any setting related 5ghz also breaks it, unloading and loading driver doesnt fix anything. I also see bunch of frequencies in the list that this chipset doesnt support (like DFS frequencies) using anything more than channel 48 also doesnt seem to work. I dont remember if this was the case all along so I am just putting it out there.

In the end I gave up manually setting frequencies, channel, transmit power and country code and left all on driver default and auto. 

I just updated to 24.10.5, i will update if anything changes.

blocktrron, 2026-01-20:
@weiqi-chen I did run in the same issue and discovered after the hardware reset the queues are never marked as active.

I've sent this patch upstream:
https://patchwork.kernel.org/project/linux-wireless/patch/20251129023904.288484-1-mail@david-bauer.net/

weiqi-chen, 2026-01-21:
I upgraded the firmware to OpenWrt 24.10.5 r29087-d9c5716d1d, and it seems the problem was solved. 
I'm not sure whether the issue disappeared because I used a configuration closer to the default settings.
I will post an update if needed.

blocktrron, 2026-01-21:
Did you observe the firmware crashing? The issue itself shall only happen on firmware crash and we've seen a 100% failure rate in this case. 

Admittedly, the cause is somewhat correlated with AP load and can most easily reproduced in busy environments (30+ STAs)

weiqi-chen, 2026-01-31:
> > [@weiqi-chen](https://github.com/weiqi-chen) please test with the linked patch, the firmware recovery will always end up in stuck tx queues, no matter the radio configuration.
> 
> I managed to compile OpenWRT (tag: OpenWrt 24.10.5 r29087-d9c5716d1d) from source, then modify the source code in `./build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-6.6.119/drivers/net/wireless/mediatek/mt76/mt76x02_mmio.c`, then run `make package/kernel/mt76/compile`, and finally reinstall these packages:
> 
> ```
>  opkg install --force-reinstall \
> >   kmod-mt76-core_6.6.119.2025.11.06~eb567bc7-r1_mipsel_24kc.ipk \
> >   kmod-mt76x02-common_6.6.119.2025.11.06~eb567bc7-r1_mipsel_24kc.ipk \
> >   kmod-mt76x2-common_6.6.119.2025.11.06~eb567bc7-r1_mipsel_24kc.ipk \
> >   kmod-mt76x2_6.6.119.2025.11.06~eb567bc7-r1_mipsel_24kc.ipk
> ```
> 
> Is there any change in the log that indicates my modification was applied? [@blocktrron](https://github.com/blocktrron)
> 
> There are no problems after reboot. Let me see if the issue happens again in the future.

No good. same issue happened again.