openwrt/mt76 #974: MT7688 with OpenWrt 22.03, When both STA and AP modes are enabled, if the STA disconnects, the AP becomes undiscoverable

sup-andy, 2025-04-22:
Device: HLK-7688A
OS: openwrt-22.03

I enabled both STA and AP modes. When the STA is connected to the upstream AP, the AP's SSID can be discovered. However, once the STA disconnects from the upstream AP, the AP's SSID also becomes undiscoverable.

Here is my configuration.

`config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ifname 'ra0'
        option ssid 'MKGW-LW-1275'
        option encryption 'psk-mixed+tkip+ccmp'
        option key '12345678'

config wifi-iface 'wifinet1'
        option device 'radio0'
        option network 'wwan'
        option mode 'sta'
        option ifname 'apcli0'
        option ssid 'aaa-iPhone'
        option disabled '0'
        option encryption 'psk2+ccmp'
        option key '12345678'`

When I turn off the "aaa-iPhone" hotspot, the following log appears on the terminal

[image]

and then the "MKGW-LW-1275" AP becomes undiscoverable.

lukasz1992, 2025-04-22:
Check 24.10.1 release

andburt-tlt, 2025-04-23:
@sup-andy , this is the way mac80211 and hostapd/wpa_supplicant works on openwrt, when sta+ap is configured, and station mode interface is not connected to external AP, device AP is down. When station connects to external AP, device AP starts on the same channel that station/external AP is using

lukasz1992, 2025-04-23:
- use two radios - one for STA second for AP
- use Mesh or WDS and not AP/STA mix
- you can try to set static channel to both STA and AP interfaces