openwrt/mt76 #976: TL-WR841N (v13) wi-fi abnormally slow

wenqingj, 2025-04-30:
I have a TL-WR841N (v13) router that runs the latest OpenWRT version 24.10.1. Its wi-fi clients, connected at link speed 72/72 Mbps (receive/transmit, b for bit) but can only download at max 2.6 MBps (B for byte). However if I attach a 802.11n AP (access point) to the router via Ethernet, and the same client connects to the AP at the same speed of 72/72 Mbps, the download is much faster, with max over 5 MBps.

Here is the details of the two tests:

(1) A laptop connects to the router via 802.11n wi-fi. The router claims to be 300 Mbps, but I guess it's because it has two antennae, and can double the channel width from 20 MHz to 40 MHz, thus quadruples the bandwidth - that is, 72 Mbps x 4 roughly equals to 300 Mbps. However in this test the router channel width is set to 20 MHz only, and I guess my laptop has only 1 antenna, hence the link speed is 72 Mbps (I checked the link speed reported by the OS which is Windows 11, and it reports 72/72 Mbps). At the same time another computer is connected to the router via an Ethernet wire (100 Mbps). The laptop runs an SFTP tool to download a large file from the other computer - the file data goes from the other computer via the Ethernet to the router, and then from the router to the laptop via the wi-fi link. There are no other computers using the router during the test. I'd expect that the download speed to be about 9 MBps (a byte has 8 bits so 72 Mbps is equivalent to 9 MBps). I understand there are some overhead cost, hence it would be somewhat lower than 9 MBps. However the test result is 2.0 Mbps ~ 2.6 MBps during the whole download.

(2) I connect a 802.11n AP device to the router via Ethernet, and then have the laptop connect to the AP (which has a channel width of 20 MHz). Again I checked the link status and confirmed that it's 72/72 Mbps. I then do a large file download. The file data goes from the other computer to the router (via Ethernet connection), then from the router to the AP device (via Ethernet connection), and finally from the AP device to the laptop (via wi-fi). However this time the speed is 4.0 MBps ~ 5.2 MBps during the whole download.

So, it seems that the 802.11n wi-fi inside the router (TL-WR841N) is abnormally slow. It might not related to this particular version of OpenWRT because I don't remember it was any faster before when using version 23.05.x. But what might be the reason for this slowness?

wenqingj, 2025-05-03:
(3) It occurred to me that I should also test out the speed in another direction. Hence I did it, sending a file from the laptop to the other computer. In this case file data streams from the laptop to the router via wi-fi, and from the router to the other computer via Ethernet connection. The speed is mostly above 5 MBps during the transfer, although before the test I checked the wi-fi connection is 72/72 Mbps (means upstream and downstream links are of same bandwidth).

This reminds me of a bug of this class of routers, which use MT7628, that had been fixed in August 2023:
https://github.com/openwrt/mt76/issues/793

That fix disabled MT7628's A-MSDU (without disabling it this class of routers would have wi-fi connection issues after some usage). A-MSDU is designed (if I read it right) to improve outstream speed. That would explain well my observation here: in test (2) the file data goes into the router via ethernet and goes out via wi-fi - with A-MSDU disabled on the 2nd leg and hence speed is reduced. While in (3) the data streams out of the laptop via wi-fi into the router, and laptop A-MSDU is not disabled hence the speed is normal (data then streams out of the router is via the Ethernet, hence irrelevant of A-MSDU).

This is not a direct evidence but it explains quite well...

wenqingj, 2025-05-04:
I managed to put stock firmware back to the router (TL-WR841N v13) and redid test (1), and the speed is fast ( > 5MBps most of the time, sometimes > 6MBps), and transferred a file bigger than 10GB without any problem. Hence it proves that the hardware is able to sustain that speed, and that OpenWRT's fix of disabling A-MSDU is probably an overkill.
The stock firmware is the latest (published on 2023-11-28) in the following page:
https://www.tp-link.com/ca/support/download/tl-wr841n/v13/#Firmware

Djfe, 2025-05-13:
how fast is OpenWrt 23.05 on said device for you?

wenqingj, 2025-05-16:
> how fast is OpenWrt 23.05 on said device for you?

It was normal speed (like 5MBps or more, on a 72Mbps link) but the wi-fi connection will be lost after some use, and to get the wi-fi back you usually need to reboot the router. The wi-fi issue was later resolved (in 23.05.5?) by disabling A-MSDU. If I remember right the speed is about halved after that.