openwrt/mt76 #979: [feature request] mt7916: allow a per-card enable_6ghz parameter

jpsollie, 2025-05-21:
I have 2 mt7916 chips in my AP,
but unless I add the parameter "enable_6ghz=y" in my /etc/modules.conf,
I cannot use the 6ghz feature.
but, when turning on this feature,
both cards seem to be locked to 6ghz, and 5ghz operation no longer works.
this is a pity, as I wanted 2 bands on 2.4ghz with limited bandwidth, one 5ghz band and one 6ghz band.
is there any way to make the "enable_6ghz=y" parameter device specific?
thanks!

relink2013, 2025-07-12:
I have to add my two cents in on this because my situation is identical. I bought two mt7916's for this exact use case and was blown away that it wasn't an option. 

But to make matters worse, apparently adding `"enable_6ghz=y"` in `/etc/modules.conf,` doesn't seem to work anymore either. It breaks the radio settings page in Luci, and even when configured manually it doesn't actually work.

jfranzlx, 2025-09-04:
@AsiaRF-Support