openwrt/mt76 #1009: mt7915: wifi unresponsive - backlog fills

maurerle, 2025-09-30:
It seems, that the same behavior as described by @ptpt52 in https://github.com/openwrt/mt76/issues/518#issuecomment-817173322 does not only affect mt7615 but also mt7915e.

```
root@ffac:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/xmit-queues
     queue | hw-queued |      head |      tail |
      MAIN |         1 |        13 |        12 |
     MCUWM |         0 |        20 |        20 |
     MCUWA |         0 |         9 |         9 |
   MCUFWDL |         0 |        51 |        51 |
root@ffac:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/rx-queues
     queue | hw-queued |      head |      tail |
         0 |      1535 |       626 |       627 |
         1 |       511 |        19 |        20 |
         2 |       511 |       218 |       219 |
         3 |      1535 |      1107 |      1108 |
         4 |       511 |       511 |         0 |
         5 |       511 |        71 |        72 |
```

`iw phy phy1 get txq` shows that there are often a lot of packages in Backlog.
Sometimes, the backlog is cleared after a few minutes, but in busy environments this takes far too long and gets the system into an unresponsive state.
Pings are not responend, trafffic is stalled.

I experienced these issues on a Zyxel NWA 55 AXE as well as on an Acer Vero W6M.
This happens on 5GHz as well as on 2.4GHz ifaces, depending on where the higher load is.
Other devices with mt7915 like COVR-X1860 and DAP-X1860 are not affected on openwrt-24.10

Workarounds:
* It helps to restart the wifi iface using `wifi`
* It does help to run `echo 7 > /sys/kernel/debug/ieee80211/phy0/mt76/sys_recovery`

What does not help:
* other recovery modes (1-6) did not help
* turning off either 2.4GHz or 5GHz did not help
* turning off the additional mesh ifaces did not help

castiel652, 2025-10-02:
I wonder if it's related to tx token.
Currently both bands shares the token instead of using per-band token.

maurerle, 2025-10-02:
@castiel652 I tried running the NWA55AXE on 2.4GHz only and still had these issues boil up.
When enabling both bands, the `Backlog` counter on 2.4GHz is nearly always 0, but in 5GHz high counts are shown.
When enabling only 2.4GHz, the counter on 2.4GHz increases >100 as well.

Here is a proper pcap share monitoring a situation of wifi instability:
https://fh-aachen.sciebo.de/s/xDJZx5QFTYW6kJD
Using `(ipv6.addr == fdac::ac) && (_ws.col.protocol == "ICMPv6")` one can see that pings to this node sometimes took more than a second.

maurerle, 2025-10-20:
After further investigation, the problem is that packets to clients which did not disassociate properly are not removed properly from the queue.
Instead, they are resent every few seconds until the hostapd removes the STA finally after 300s of inactivity.
This leads to stalled tx in environments with many clients which move out of range without being able to send a disconnect - typically in outdoor environments.

## Failure state reproduction

The issue can be removed by shutting of a client without a disconnect. This is not possible with `rmmod` or `rfkill block wlan`, but `poweroff -f` does the job.
Small delays in the pings can be seen with that.
If multiple STA are in this situation, the problem becomes more visible.

## Symptoms

Pings to other (even local) ip addresses can increase from 20ms to >1000ms during a stalled tx event.

Such situations are come with stalled hw-queues for all stations.
This can be seen as the per STA counters in `cat /sys/kernel/debug/ieee80211/phy1/mt76/hw-queues` (or similar for phy0) do not change between subsequent calls.

## Known workarounds

It helps to disconnect such stations using `iw dev $dev station del $mac` - after removal the stations tx queues are unstuck again.

The script from https://github.com/openwrt/mt76/issues/922#issuecomment-2525416952 works to solve this by disconnecting stations which do not respond.

But a much better approach is to set `option max_inactivity 30` in /etc/config/wireless for the respective ifaces.
This sets the hostapd config `ap_max_inactivity` https://openwrt.org/docs/guide-user/network/wifi/basic#inactivity_timeout_options
This option is preferred as it first checks if the device still responds at all if nothing was sent - the default interval is 300s.

rany2, 2026-04-18:
@csharper2005 I use these patches from Gluon:

- https://github.com/rany2/openwrt/commit/3fdd68f108066d8427be5ce804523cebee52fd68
- https://github.com/rany2/openwrt/commit/8ce4a905b9d0ee06480b7fabde0a424c013e794a
- https://github.com/rany2/openwrt/commit/06324dca7778ed98e7b64abd12d606658042a47e
- https://github.com/rany2/openwrt/commit/6b3c6649cfa4f2f5d8f814f849ee3d99b4e6a979
- https://github.com/rany2/openwrt/commit/55185fa2f89e554798d9836c6e1525ba84752b50

It seems to be working well.